riscvarchive/riscv-binutils-gdb #180: How to do abstract memory access by using gdb?

thomascp, 2019-10-09:
Hi,

In our board, there is a module mapped into CPU, but not mapped into the system bus.
So I can not access the registers of this module by using 'x address' in gdb.
Our hardware guy told me, 'General philosophy here is to use abstract memory commands to get  to the internal core memories (DCCM/ICCM/internal module) using the debugger.'.
I read some parts of riscv debug spec, but get nothing about how to execute abstract memory commands in gdb.
Could anyone help me here? Thanks!

BR,
Peng

timsifive, 2019-10-09:
Generally speaking gdb doesn't have a concept of different kinds of memory accesses (although it can be aware of where flash is).
In theory you could send a command to the debug monitor to tell it what kind of memory accesses to use from now (or even fancier, for certain address ranges) but OpenOCD doesn't have support for that either. The closest we have is "riscv set_prefer_sba" but that doesn't do what you want. It would make sense to modify and rename that command so it lets you specify the order that memory accesses should be tried in (progbuf, sba, and abstract).

timsifive, 2019-10-10:
There is no code in gdb. This is all in OpenOCD: https://github.com/riscv/riscv-openocd/blob/9aac179cf28fe2af49efd6eeccc6cfcea9b2db3b/src/target/riscv/riscv.c#L1915

You can file an issue there, but I won't have time to get to this in quite a while. If you want to put together a pull request I'll be happy to review it.