riscvarchive/riscv-binutils-gdb #181: how can i automatic  allocate register when i use vector instruct？

chinadirk, 2019-10-11:
My code :
     long long a = 5,b = 2,c = 6;
	
	asm("vwmul.vx %dest, %src1, %src2"
	    :[dest]"=v"(a)
		:[src1]"v"(b),[src2]"r"(c));
compile error : 
main.c:17:2: error: impossible constraint in 'asm'
  asm("vwmul.vx %dest, %src1, %src2"

jim-wilson, 2019-10-11:
There is no compiler support as yet, so no way to write extended asms.  You can only write assembly language code.

chinadirk, 2019-10-12:
thank you for your early feedback.
what time do it can be used,please?

jim-wilson, 2019-10-12:
There is no dedicated person working on it, so it is impossible to say when it might be ready, other than the fact that it is unlikely to be anytime soon.

closed 2019-10-12 by chinadirk

wzgpeter, 2021-03-09:
is there any progress on this issue. thank you!