riscvarchive/riscv-binutils-gdb #189: Assembler: Branch Instructions

musicist288, 2019-11-27:
I'm running into a strange issue issue with the riscv assembler related to branch instructions. 

I just compiled the toolchain which built the assembler from this commit: d91cadb45f3ef9f96c6ebe8ffb20472824ed05a7`

I wrote a toy program with some branch instructions and ran:
`riscv64-unknown-elf-as -mabi=ilp32 -march=rv32i  branch-test.s -o branch-test.elf`

on

`branch-test.s`
```
main:
    li      a0,1
    li      a1,-1
    li      a3,1
    li      a4,2
    bne     a0,a3,1
    beq     a0,a1,1
    bge     a1,a0,1
    bgeu    a3,a4,1
    blt     a4,a3,1
    bltu    a1,a0,1
    li      ra,5
```

Then I disassembled the file and got the following output:

`/opt/riscv/bin/riscv64-unknown-elf-objdump -S branch-test.elf`
```
branch-test.elf:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
main():
   0:	00100513          	li	a0,1
   4:	fff00593          	li	a1,-1
   8:	00100693          	li	a3,1
   c:	00200713          	li	a4,2
  10:	00d50463          	beq	a0,a3,18 <main+0x18>
  14:	0000006f          	j	14 <main+0x14>
  18:	00b51463          	bne	a0,a1,20 <main+0x20>
  1c:	0000006f          	j	1c <main+0x1c>
  20:	00a5c463          	blt	a1,a0,28 <main+0x28>
  24:	0000006f          	j	24 <main+0x24>
  28:	00e6e463          	bltu	a3,a4,30 <main+0x30>
  2c:	0000006f          	j	2c <main+0x2c>
  30:	00d75463          	bge	a4,a3,38 <main+0x38>
  34:	0000006f          	j	34 <main+0x34>
  38:	00a5f463          	bgeu	a1,a0,40 <main+0x40>
  3c:	0000006f          	j	3c <main+0x3c>
  40:	00500093          	li	ra,5
```

It looks to me that the branch instructions got swapped. For example, the instruction at 0x10 disassembled to `beq` when in the source file, it was `bne`. Is this an issue with the assembler/disassembler of is there something I'm doing that's causing this issue?

aswaterman, 2019-11-27:
The linker thinks that the branch target is far away, and so instead of a
simple branch, it emits a branch around a jump, which has further reach.
If you make the branch target something nearby (e.g. locally declare a
label and branch to that), you'll see what you expect.

musicist288, 2019-11-28:
Perfect! Thank you! Got the output I was expecting when making those changes.

closed 2019-11-28 by musicist288