riscvarchive/riscv-binutils-gdb #194: How can I use this in riscv32-based virtual machine?

geonheec, 2020-01-10:
Hi, can I cross compile this gdb to riscv32 executable program?

I think there's two way, [1] compile this repository with riscv32-gcc in x86_64 based machine and use the executable gdb binary in my riscv32-based virtual machine or [2] clone this repo at riscv32-based virtual machine and compile in that vm.

I wonder if either is possible.

If possible, how can I modify the Makefile or configure file to output riscv32-executable program??

kito-cheng, 2020-01-10:
Second one would be easier if you are using some RISC-V linux distribution like Fedora or debian, the configure option just give `./configure --prefix=<install-path>`, configure script will auto detect you are using RISC-V host and then produce right Makefile :)

jim-wilson, 2020-01-10:
If you want to cross compile a binary to run on a riscv linux system, then you must use a C library and kernel headers that are compatible with the target.  The stuff in riscv-gnu-toolchain is unlikely to be compatible with the target.  I would not recommend using riscv-gnu-toolchain for this.  I would suggest instead that you create a sysroot from the target, where sysroot includes the libraries and headers.  Then build a cross compiler using that sysroot.  Then use that cross compiler to build a gdb that can run on the target.  If you have never done this before, it may take some time to learn how to do it correctly.

Building gdb natively is a much easier option if you already have a riscv linux system, though a gdb build on riscv-linux running on qemu may take a while.

And you don't need riscv/riscv-binutils-gdb to do any of this.  You can use the FSF GDB sources.  All of the linux distros are using the FSF GDB sources directly.

Speaking of which, if you have a distro, then all of them already have a gdb binary which you can just apt install or whatever.  But you mentioned riscv32-linux, so maybe you are trying to build your own via some other way, such as via buildroot.  I don't know if buildroot supports building a gdb to run on the target, but if it does, that might be even easier third option.

We don't have gdbserver yet unfortunately but it is being worked on.

geonheec, 2020-01-14:
Thank you so much.
I just find other riscv32-linux image which supports gdb.

Have a nice day!

closed 2020-01-14 by geonheec