riscvarchive/riscv-binutils-gdb #196: GDB RISCV V support

RaphaelKlink, 2020-01-14:
Is there a current or planned support for the RISCV Vector Extension for the gdb part?

jim-wilson, 2020-01-14:
No.  Someone will eventually volunteer to do the work.

fabianpedd, 2021-09-29:
Have there been any updates since the issue was first opened?

jim-wilson, 2021-09-29:
FSF GDB sources have basic support for the vector registers.  This was added in June and is in the GDB 11 release.  It probably only works with qemu and/or openocd at this time.

We are still lacking ptrace support to read/write vector registers.  So a gdb running on a vector linux system won't be able to read/write vector registers.  This probably won't happen until people have vector capable hardware running vector capable linux distros.

There will likely be more changes needed after this.  For instance, if vector registers are added to ucontext_t, then the signal handler debug support will need support to read vector registers from ucontext_t.  The core file debug support will need to be updated to read vector registers from the core dump.  Though we need OS support to put them in the core dump file first.  The gdbserver port will need vector register support added to it.  Etc.

fanghuaqi, 2022-01-04:
Is there a branch in this repo to evaluate the feature of support vector register read or write in gdb?

jim-wilson, 2022-01-04:
GDB development happens upstream.  See
    https://sourceware.org/gdb/
and use the master branch sources.  But vector support is rather limited at this time as I mentioned.

If you want to use this with riscv-gnu-toolchain, you can use the --with-gdb-src= configure option, or you can use git tricks (e.g. git remote).

fanghuaqi, 2022-01-04:
Thanks for sharing this info