riscvarchive/riscv-binutils-gdb #203: The immediate range of alias instructions in the test set

joy2myself, 2020-02-29:
File: riscv-binutils-gdb/gas/testsuite/gas/riscv/vector-insns.s
Line:1242, 1246, 1250, 1254
```
vmsltu.vi v4, v8, -15
vmsgeu.vi v4, v8, -15
vmsltu.vi v4, v8, -15, v0.t
vmsgeu.vi v4, v8, -15, v0.t
```
The range of the unsigned immediate in this four instructions is 1 to 16 according to v0.8 spec.
The test case for lower bound should be 
```
vmsltu.vi v4, v8, 1
vmsgeu.vi v4, v8, 1
vmsltu.vi v4, v8, 1, v0.t
vmsgeu.vi v4, v8, 1, v0.t
```

aswaterman, 2020-02-29:
The assembler is getting this right.  The standard RISC-V assembly syntax accepts signed immediates even when the immediates represent unsigned quantities.  For example, in the scalar RV32 assembly syntax, you can express the computation `a0 = a0 < 0xfffffffe` by writing `sltiu a0, a0, -2`.

The NOTE in the V spec you're referring to is just trying to explain what the immediate range represents; it's not dictating the assembly syntax.

joy2myself, 2020-02-29:
Thank you for your reply. I'm implementing an Assembler for rvv-llvm. Should I use signed operand for these immediates if I want to pass these test cases?

aswaterman, 2020-02-29:
Yeah, I would recommend following suit.

Nelson1225, 2020-07-13:
I think this PR is the same as the one mentioned in the riscv-v-spec closed issue (https://github.com/riscv/riscv-v-spec/issues/327).  The riscv-v-spec issue should be resolved, so I prefer to close this one, too.

You can reopen this PR or create a new one if needed.  Thanks.

closed 2020-07-13 by Nelson1225