riscvarchive/riscv-binutils-gdb #206: Strange interpret of pseudo instruction vmsltu.vi

HanKuanChen, 2020-04-01:
I am using `rvv-0.9.x` branch.

I have a `.S` which contains `vmsltu.vi v25, v14, 0`.
`v25` should be `0` for all elements.
But the assembler interprets this instruction into `vmsleu.vi v25,v14,-1`, which makes `v25` be `1` for all elements.

aswaterman, 2020-04-01:
Options are:

- Reject the pseudoinstruction when imm=0.
- Convert the pseudoinstruction to `vmsne.vv v25, v14, v14` when imm=0, which has the correct effect of setting all the elements to 0 (and also it maintains any illegal-instruction trapping behavior because of the register numbers).

kito-cheng, 2020-04-01:
> Options are:
> 
> * Reject the pseudoinstruction when imm=0.
> * Convert the pseudoinstruction to `vmsne.vv v25, v14, v14` when imm=0, which has the correct effect of setting all the elements to 0 (and also it maintains any illegal-instruction trapping behavior because of the register numbers).

I prefer second option, because it's pseudoinstruction, it possible to make it by another instruction, so why not?

@Nelson1225

Nelson1225, 2020-04-01:
Should we convert the vmsgeu.vi with zero imm to vmsne.vv too?

aswaterman, 2020-04-01:
The spec says "Note, vmsgeu.vi with immediate 0 is not useful as it is always true" -- I think Krste was implying that "there's no reason to provide a pseudo-op, because the instruction isn't useful."

I don't feel strongly one way or the other, but we should be consistent between vmsltu and vmsgeu (make both illegal with 0 operation, or convert both to vmsne/vmseq)

Nelson1225, 2020-04-01:
> The spec says "Note, vmsgeu.vi with immediate 0 is not useful as it is always true" -- I think Krste was implying that "there's no reason to provide a pseudo-op, because the instruction isn't useful."
>
> I don't feel strongly one way or the other, but we should be consistent between vmsltu and vmsgeu (make both illegal with 0 operand, or convert both to vmsne/vmseq)

Agree, two method are good to me.  I will create a riscv-v-spec issue to make sure which way is preferred by team :)

Thanks

Nelson1225, 2020-04-08:
@HanKuanChen, This should be fixed in the rvv-0.9.x branch.  Thanks :)

closed 2020-04-09 by HanKuanChen