riscvarchive/riscv-binutils-gdb #208: Wrong expantion for vmsge.vx and vmsgeu.vx

kito-cheng, 2020-04-07:
vmsge.vx and vmsgeu.vx will use extra register which not appear in the operand list, and not defined any where.

Testcase:
```
test:
        vmsge.vx v4, v8, a1
        vmsgeu.vx v4, v8, a1
```
Objdump:
```
x.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <test>:
   0:   6e85c257                vmslt.vx        v4,v8,a1
   4:   7685a257                vmnand.mm       v4,v8,v11
   8:   6a85c257                vmsltu.vx       v4,v8,a1
   c:   7685a257                vmnand.mm       v4,v8,v11
```

Command:
`$ riscv64-unknown-elf-as x.s -o x.o  -march=rv64gcv`
`$ riscv64-unknown-elf-objdump -d x.o`

Nelson1225, 2020-04-07:
I think they should be expanded to

00000000 <.text>:
   0:   6e85c257                vmslt.vx        v4,v8,a1
   4:   76422257                vmnot.m v4,v4
   8:   6a85c257                vmsltu.vx       v4,v8,a1
   c:   76422257                vmnot.m v4,v4

And `vmnot.m v4, v4` = `vmnand.mm v4, v4, v4`.  I will create a PR after running the tests.

Thanks

Nelson1225, 2020-04-08:
This should be fixed with https://github.com/riscv/riscv-binutils-gdb/issues/206 in rvv-0.9.x branch :)

Nelson1225, 2020-04-21:
https://github.com/riscv/riscv-binutils-gdb/pull/210

kito-cheng, 2020-04-21:
Fixed via  #207

closed 2020-04-21 by kito-cheng