riscvarchive/riscv-binutils-gdb #217: Missing CSR: tinfo, tcontrol, mcontext and scontext

hauhsu, 2020-05-28:
[RISC-V debug spec](https://riscv.org/specifications/debug-specification/) defines some trigger registers (section 5.2). And these registers should be accessible using `csr` instruction.
But I have the following assembly code that triggers unknown CSR error:
```
$ cat test.s
.text
    csrr t0, tdata1   # pass
    csrr t0, tinfo    # error
    csrr t0, tcontrol    # error
    csrr t0, mcontext    # error
    csrr t0, scontext    # error
$ riscv32-elf-as test.s
test.s: Assembler messages:
test.s:3: Error: unknown CSR `tinfo'
test.s:4: Error: unknown CSR `tcontrol'
test.s:5: Error: unknown CSR `mcontext'
test.s:6: Error: unknown CSR `scontext'
```

I think these registers are missed in _binutils/include/opcode/riscv-opc.h_?:
```
...
#define CSR_TSELECT 0x7a0
#define CSR_TDATA1 0x7a1
#define CSR_TDATA2 0x7a2
#define CSR_TDATA3 0x7a3
#define CSR_DCSR 0x7b0
#define CSR_DPC 0x7b1
#define CSR_DSCRATCH 0x7b2
...
```

jim-wilson, 2020-05-28:
We weren't aware that there were debug csrs that we needed to support.  We will have to put this on our todo list.  Meanwhile, you can access the csrs by using their address.

timsifive, 2020-05-29:
User beware that if you tell gdb to set a hardware breakpoint, that's likely to overwrite whatever you wrote to those registers.

However, there are a lot of possible hardware triggers that gdb has no model for, so sometimes writing these register values yourself is the only way to do what you want.

jim-wilson, 2020-05-29:
@Nelson1225 can you work on this?

Nelson1225, 2020-05-30:
OK I will take care of this :)

hauhsu, 2020-06-08:
Thank you guys 😄 .

Nelson1225, 2020-06-30:
Sorry for the late reply.  I didn't add the missing debug CSR until recently, since there are some related problems need to be resolved first.  The current FSF binutils should use the missing debug CSR mentioned above now.  Please feel free to let me know if it is good to you.  Thanks.

Nelson1225, 2020-07-13:
The missing debug CSR should be added on the FSF binutils.  You can reopen this PR or create a new one if needed.  Thanks.

closed 2020-07-13 by Nelson1225