riscvarchive/riscv-binutils-gdb #221: branches of vector extension are not merged into mainline

rjiejie, 2020-07-08:
Hi,

  Some of vector spec versions are stable for using in productions as official says :
"
This is a draft of a stable proposal for the vector speci cation to be used for implementation and evaluation. Once the draft label is removed, version 0.7 is intended to be stable enough to begin developing toolchains, functional simulators, and initial implementa- tions, though will continue to evolve with minor changes and updates.
" 👍 

  I can not find any code in the mainline :(
  Is there any plan to merge these branches into binutils mainline ?

kito-cheng, 2020-07-08:
Upstream/FSF GNU toolchain intend to support ratified extension only, e.g. extension with v1.9 or v2.0. in order to prevent multiple incompatible extension version on upstream.

jim-wilson, 2020-07-08:
This question was asked in 2 places.  I gave an answer in the other place.

closed 2020-07-08 by jim-wilson

jim-wilson, 2020-07-08:
Make that 3 places.  Sigh.  Getting 3 answers from 3 different people, fortunately all basically the same answer.  Please stop asking the same question in multiple places.  Pick one place to have a discussion.