riscvarchive/riscv-binutils-gdb #225: kallsyms failure with riscv-ld 2.34

Phantom1003, 2020-08-09:
Hi, these days I upstream my toolchain to gcc 10.1 and ld 2.34, but I found when I compile linux with clang, following error will be thrown:
```
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
  KSYM    .tmp_vmlinux.kallsyms1.o
kallsyms failure: relative symbol value 0xffffffe000000000 out of range in relative mode
/home/phantom/PPProject-clang/linux/Makefile:1139: recipe for target 'vmlinux' failed
make[2]: *** [vmlinux] Error 1
```

After check I found the bugs should from ld, cause I can got vmlinux with riscv-ld 2.32, can you generally infer where this bug come ?

kito-cheng, 2020-08-10:
Do you mind describe more detail about this? e.g. kernel config, kernel version, how to build?

Phantom1003, 2020-08-10:
> Do you mind describe more detail about this? e.g. kernel config, kernel version, how to build?
Thanks, I use llvm-project(release/11.x) amd linux(linux-5.8.y), I use 5.8 because it is the first runnable version of vmlinux compiled with clang, the config I use is base from Freedom U SDK:
```
CONFIG_SMP=y
# CONFIG_COMPACTION is not set
CONFIG_HZ_100=y
CONFIG_PCI=y
CONFIG_PCI_DEBUG=y
CONFIG_PCIE_XILINX=y
CONFIG_CROSS_COMPILE="riscv64-unknown-linux-gnu-"
CONFIG_DEFAULT_HOSTNAME="ucbvax"
# CONFIG_CROSS_MEMORY_ATTACH is not set
CONFIG_NAMESPACES=y
CONFIG_BLK_DEV_INITRD=y
# CONFIG_RD_BZIP2 is not set
# CONFIG_RD_LZMA is not set
# CONFIG_RD_XZ is not set
# CONFIG_RD_LZO is not set
# CONFIG_RD_LZ4 is not set
CONFIG_EMBEDDED=y
# CONFIG_BLK_DEV_BSG is not set
CONFIG_PARTITION_ADVANCED=y
# CONFIG_IOSCHED_DEADLINE is not set
CONFIG_NET=y
CONFIG_PACKET=y
CONFIG_UNIX=y
CONFIG_INET=y
# CONFIG_INET_XFRM_MODE_TRANSPORT is not set
# CONFIG_INET_XFRM_MODE_TUNNEL is not set
# CONFIG_INET_XFRM_MODE_BEET is not set
# CONFIG_INET_DIAG is not set
# CONFIG_IPV6 is not set
CONFIG_DEVTMPFS=y
CONFIG_DEVTMPFS_MOUNT=y
# CONFIG_FIRMWARE_IN_KERNEL is not set
CONFIG_MTD=y
CONFIG_MTD_BLOCK_RO=y
CONFIG_MTD_M25P80=y
CONFIG_MTD_SPI_NOR=y
CONFIG_VIRTIO_BLK=y
CONFIG_EEPROM_AT24=y
CONFIG_BLK_DEV_SD=y
CONFIG_ATA=y
CONFIG_SATA_AHCI=y
CONFIG_SATA_SIL24=y
CONFIG_NETDEVICES=y
CONFIG_VIRTIO_NET=y
CONFIG_MACB=y
CONFIG_MICROSEMI_PHY=y
# CONFIG_INPUT_KEYBOARD is not set
# CONFIG_INPUT_MOUSE is not set
# CONFIG_VT is not set
CONFIG_SERIAL_8250=y
CONFIG_SERIAL_8250_CONSOLE=y
# CONFIG_SERIAL_8250_PCI is not set
CONFIG_SERIAL_OF_PLATFORM=y
CONFIG_SERIAL_SIFIVE=y
CONFIG_SERIAL_SIFIVE_CONSOLE=y
CONFIG_HVC_RISCV_SBI=y
CONFIG_VIRTIO_CONSOLE=y
# CONFIG_HW_RANDOM is not set
CONFIG_I2C=y
CONFIG_I2C_OCORES=y
CONFIG_SPI=y
CONFIG_SPI_SIFIVE=y
CONFIG_GPIOLIB=y
CONFIG_GPIO_SYSFS=y
CONFIG_GPIO_SIFIVE=y
CONFIG_POWER_RESET=y
CONFIG_POWER_RESET_GPIO_RESTART=y
CONFIG_USB=y
CONFIG_USB_XHCI_HCD=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_STORAGE=y
CONFIG_MMC=y
CONFIG_MMC_SPI=y
CONFIG_NEW_LEDS=y
CONFIG_LEDS_CLASS=y
CONFIG_LEDS_PWM=y
CONFIG_LEDS_TRIGGERS=y
CONFIG_LEDS_TRIGGER_MTD=y
CONFIG_LEDS_TRIGGER_HEARTBEAT=y
CONFIG_LEDS_TRIGGER_PANIC=y
CONFIG_VIRTIO_MMIO=y
CONFIG_CLK_U54_PRCI=y
CONFIG_CLK_GEMGXL_MGMT=y
# CONFIG_IOMMU_SUPPORT is not set
CONFIG_PWM=y
CONFIG_PWM_SIFIVE=y
CONFIG_NVMEM_SIFIVE_U500_OTP=y
CONFIG_EXT3_FS=y
# CONFIG_PROC_PAGE_MONITOR is not set
CONFIG_TMPFS=y
# CONFIG_MISC_FILESYSTEMS is not set
# CONFIG_NETWORK_FILESYSTEMS is not set
CONFIG_CMDLINE_BOOL=y
CONFIG_CMDLINE="earlyprintk"
CONFIG_PRINTK_TIME=y
CONFIG_STRIP_ASM_SYMS=y
CONFIG_DEBUG_SECTION_MISMATCH=y
CONFIG_STACKTRACE=y
# CONFIG_CRYPTO_HW is not set
```

kito-cheng, 2020-08-11:
Hmmm.... I can't reproduce that with llvm/clang 11 + linux kernel v5.8(bcf876870b95592b52519ed4aafcf9d95999bc9c) + GNU ld (GNU Binutils) 2.34(d7f734bc7e9e5fb6c33b433973b57e1eed3a7e9f)

Phantom1003, 2020-08-11:
Sorry I miss some information, after check, we also need to set some variables during linux make 

```
make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv \
		CC=clang KBUILD_CFLAGS_KERNEL="-g" LLVM_IAS=1 \
		vmlinux
```
The error only happen when both KBUILD_CFLAGS_KERNEL and LLVM_IAS are set

kito-cheng, 2020-08-11:
Thanks your info, I can reproduce now.

closed 2021-08-29 by Phantom1003