riscvarchive/riscv-binutils-gdb #234: How gdb register csr

Phantom1003, 2020-10-15:
I'm trying to implement a simple gdb server to support a interfaces above qemu, now I can use gdb serial protocol to communicate with qemu, to get general register and memory. But I failed to get the value in csr, for example I want the value in mtvec, I send $pa5#xx, and I got $#00, and I filter the packages send by gdb, I found that gdb will request some xml files, is this necessary step to register csr ?

jim-wilson, 2020-10-15:
Yes, you must specify the register set with xml files, including the available csrs.  gdb, qemu, and openocd have example xml files for RISC-V.

Phantom1003, 2020-10-17:
Thanks, for qemu we should to request a .xml to enable the `hasxmlfiles` variable with a `qXfer:feature:read` command, usually we use target.xml, then you can use `p` command to get csr value.

closed 2020-10-17 by Phantom1003