riscvarchive/riscv-binutils-gdb #243: About the opcode constraints documentation.

were, 2020-12-21:
Dear All,

I am hacking the RISCV codebase for my own research.
I am confused by the 4th column of the [opcode definition](https://github.com/riscv/riscv-binutils-gdb/blob/2cb5c79dad39dd438fb0f7372ac04cf5aa2a7db7/opcodes/riscv-opc.c#L200).
I cannot find full documentation of the letters in that column in [gcc internal](https://gcc.gnu.org/onlinedocs/gccint.pdf).
Is there any material I can learn from?

Regards,
Jian

kuanlinchentw, 2020-12-21:
Hi,

You may reference [riscv_ip](https://github.com/riscv/riscv-binutils-gdb/blob/2cb5c79dad39dd438fb0f7372ac04cf5aa2a7db7/gas/config/tc-riscv.c#L1845) or [validate_riscv_insn ](https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.35/gas/config/tc-riscv.c#L885) to know the letter definition.
If there is any document describing the opcode definition, I'd like to know it, too. XD

jim-wilson, 2020-12-21:
We don't have documentation for the operand letters.  You have to read the gas sources to see what they do.  The ones that start with C are two letter codes for compressed instructions.  The rest are one letter codes.

were, 2020-12-21:
@kuanlinchentw @jim-wilson Thank you for your prompt response. I will be navigating the codebase then.

closed 2020-12-21 by were