riscvarchive/riscv-binutils-gdb #248: riscv-binutils-2.35-rvb: `_`instead of `.` ?

rdolbeau, 2021-02-25:
Hello,

Trying to compile the branch `riscv-binutils-2.35-rvb` (with support of the B extension) as part of the riscv-bitmanip repo, the build fails on the `.' (dot) in `xperm.[nbhw]` in the first parameter of the `DECLARE_INSN` macros inside the file `include/opcode/riscv-opc.h`.

Replacing the `.` by a `_` in all four macros (as other instructions seem to do):

```
-DECLARE_INSN(xperm.n, MATCH_XPERMN, MASK_XPERMN)
-DECLARE_INSN(xperm.b, MATCH_XPERMB, MASK_XPERMB)
-DECLARE_INSN(xperm.h, MATCH_XPERMH, MASK_XPERMH)
-DECLARE_INSN(xperm.w, MATCH_XPERMW, MASK_XPERMW)
+DECLARE_INSN(xperm_n, MATCH_XPERMN, MASK_XPERMN)
+DECLARE_INSN(xperm_b, MATCH_XPERMB, MASK_XPERMB)
+DECLARE_INSN(xperm_h, MATCH_XPERMH, MASK_XPERMH)
+DECLARE_INSN(xperm_w, MATCH_XPERMW, MASK_XPERMW)
```

solve the compilation issue. Is that a proper fix, or should it be able to compile as-is?

jim-wilson, 2021-02-25:
Everything builds for me, but this is because I'm building binutils from a binutils branch and gdb from a gdb branch, and the B support is not on the gdb branch.  If I try to build gdb from the binutils branch it fails.  Note that in general building gdb from a binutils release branch is not recommended and not supported.  But this is a bug that will show up when the code is upstreamed or copied to the gdb branch and does need to be fixed.  I will take care of this.

Nelson1225, 2021-03-02:
This should be fixed in the branch `riscv-binutils-2.35-rvb` https://github.com/riscv/riscv-binutils-gdb/commit/d190acd1c264407beba39cb78c286fb35ab641e4, so I close this issue, thanks.

closed 2021-03-02 by Nelson1225