riscvarchive/riscv-binutils-gdb #250: How to add a new custom instruction?

Nado15, 2021-02-28:
Hello,

is there a current guide how to add a new RISC-V instruction to riscv-binutils-gdb?

Nelson1225, 2021-03-02:
No, we don't have any standardized guidance to introduce it.  Perhaps there are, but at least I don't know.  But the following discussion maybe help,
https://github.com/riscv/riscv-gnu-toolchain/issues/782

Nado15, 2021-03-03:
Ok thanks I checked the source and I have a question. Lets says I wanne add a instruction that increments the value in the _rd_ register. And I decide to create a U-Type similar instruction. Now I take for example custom-0 (from the table 24.1 in the riscv spec) with the opcode 0001011. Now I have to modify the riscv-opc.c

`{"addone",          32, INSN_CLASS_I,   "d",   MATCH_ADDONE,  MASK_ADDONE, match_opcode, 0 }`

So as far as I understand, now I am saying that I have added a new instruction to the I set and with the "d" I am telling the risc-v assembler that _yyyyyyyyyyyyyyyyyyxxxxx0001011_ the register is encoded in the x bits and he ignores the rest of the y bits? Would that be a valid instruction?

jim-wilson, 2021-03-05:
MASK_ADDONE is all of the non operand bits.  MASK_ADDONE plus the "d" bits should be 32 one bits.  MATCH_ADDONE is every non-operand bit that is non-zero.  So the encoding is MATCH_ADDONE plus the "d" bits.  On disassembly, if (insn & MASK_ADDONE) == MATCH_ADDONE then we print the addone opcode followed by its operands.

Nado15, 2021-03-05:
>MASK_ADDONE plus the "d" bits should be 32 one bits.

But d is not a fixed bitpattern right (like in my example with the "x" bits)? d can be one register so if in one instruction d has the encoding for x1 and in the next instruction d has the encoding for x2 how should that work out?

jim-wilson, 2021-03-05:
> > MASK_ADDONE plus the "d" bits should be 32 one bits.

I meant all of the bits in the d field.  It should be obvious that the operand bits plus non-operand bits should cover every bit.  If they don't, then there is a bug either in the mask or the operands.

Nado15, 2021-03-08:
I got it know, thank you!

> Ok thanks I checked the source and I have a question. Lets say I wanne add a instruction that increments the value in the _rd_ register by 1. And I decide to create a U-Type similar instruction. Now I take for example custom-0 (from the table 24.1 in the riscv spec) with the opcode 0001011. Now I have to modify the riscv-opc.c
> 
> `{"addone", 32, INSN_CLASS_I, "d", MATCH_ADDONE, MASK_ADDONE, match_opcode, 0 }`
> 
> So as far as I understand, now I am saying that I have added a new instruction to the I set and with the "d" I am telling the risc-v assembler that _yyyyyyyyyyyyyyyyyyxxxxx0001011_ the register is encoded in the x bits and he ignores the rest of the y bits? Would that be a valid instruction?

Btw. for the next person who reads this, this wont work as the y bits are not covered similar to #233 
Add the "u" operand and you have a correct U-Type instruction

`{"addone", 32, INSN_CLASS_I, "d,u", MATCH_ADDONE, MASK_ADDONE, match_opcode, 0 }` 

MATCH_ADDONE = 0x0000000b
MASK_ADDONE= 0x00000007f

closed 2021-03-08 by Nado15