riscvarchive/riscv-binutils-gdb #256: a tricky one about gdb

ThomasSun001, 2021-04-19:
I'm using gdb to debugger xv6 OS running on qemu, I'm about to debug it, but it seems that it lost something...
`~/xv6-labs-2020# riscv64-unknown-elf-gdb -tui
riscv64-unknown-elf-gdb: TUI mode is not supported
`
I just install all the tool in one command, it take hours, so rebuild all tools are time-consuming, is there a quick way to to fix it, say, to just add support??

jim-wilson, 2021-04-19:
tui requires curses support.  Some linux distros don't install curses by default anymore.  I think few people care about gdb tui support so it isn't in our list of build prerequisites.  The OS package you need is probably something like libncurses5-dev.

If you still have your build tree, you can cd into build-gdb-newlib and do a make and make install to just build and install gdb.  Except in this case you need to reconfigure which makes it a little more complicated.  I'm not sure the exact command.  ./config.status --recheck might work, or it might not.  Another approach is at the top of the build tree to rm stamps/build-gdb-newlib and then type make, and only gdb will be rebuilt and reinstalled.  This will rm -rf the gdb build tree first, which will ensure that it gets reconfigured.  And check the gdb config.log to make sure it found curses.  This is the part where it looks for the waddstr function.  If it doesn't find a library with waddstr, then it won't enable tui.

ThomasSun001, 2021-04-19:
thx for your support, actually I need the layout like layout split, then I can step by step debugging....gdb supports that but riscv64-unknown-elf-gdb does NOT support that, gdb-multiarch also supports that, but only riscv64-unknown-elf-gdb can work on my machine, I rebuild with enable tui, and make install, seems not to install in the right place...I'm really new to it

jim-wilson, 2021-04-20:
The riscv gdb is using FSF gdb sources, and should work the same as other gdb versions.  If you are having trouble building it, and you don't need to customize it, you can try downloading binaries.  The SiFive web site for instance has pointers to some precompiled binaries on the software page.

liudamu, 2021-12-22:
I also meet this problem with centos7 and riscv64-unknown-elf-gd with version 11.0.50.20210718-git.After installing ncurses-devel and recompling the whole riscv64-gnu-toolchains it works! Hope this will give you some advice.

tianlanlanlan, 2022-06-14:
i have the same problem on ubuntu22.04.i tryed download the libncurses5-dev and rebuild the riscv-gnu-toolchain, the gdb tui funciton is back.i upload the prebuilt riscv64-unknown-elf-gdb in [https://github.com/tianlanlanlan/prebuilt-riscv-gnu-toolchain-gdb/tree/master](url)