riscvarchive/riscv-binutils-gdb #261: ld: cannot find default versions of the ISA extension `i', when linking rv32e

kenta2, 2021-05-10:
I built riscv-gnu-toolchain (commit b715e4f01b43efef487166f75d5d85d3c33fa7ef , tag: 2021.04.23) with `--with-arch=rv32e --with-abi=ilp32e` .

Then, compiling and linking the simplest possible C program `main(){}` results in the following error when linking:

```
.../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find default versions of the ISA extension `i'
```

The problem goes away if I roll back riscv-gnu-toolchain to (commit d45cfc68be6ce0a2f69daf66e64fc446224b3416 , tag: 2021.04.07), so the problem was likely introduced in "Bump Binutils to 2.36.1".

Drilling down into riscv-binutils, the error message was added in commit  7ef19aa616aee98f0a5c7c948d0db56a9f3c5e57 to bfd/elfxx-riscv.c .

kito-cheng, 2021-05-10:
IIRC @Nelson1225 has fixed this issue on upstream? Nelson could you provide commit id for that?

Nelson1225, 2021-05-10:
Probably three commits in binutils mainline need to be applied,

1. RISC-V: Add i-ext as the implicit extension when e-ext is set.
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=f0bae2552db1dd4f1995608fbf6648fcee4e9e0c

2. RISC-V: The version of i-ext should be RISCV_UNKNOWN_VERSION when expanding g-ext.
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=f5b10973531cdda7e1aba7496f70476b5146fcf8

3. RISC-V: Don't report the mismatched version warning for the implicit extensions.
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=e3839c100fdb15d05e6a2fd44557ffdd6fb921b8

The commit ids are mentioned in the links.

Nelson1225, 2021-05-10:
@kito-cheng Should we create two more branches to track the binutils mainline and integration branch here?

Error1000, 2021-06-12:
Bump? I am still seeing this issue ...

jim-wilson, 2021-06-15:
Fixed.  Ideally we should have rv32e targets in the regression testsuite to catch problems like this, but there are too many testsuite failures for rv32e, so this will be a fair amount of work to fix.

closed 2021-06-15 by jim-wilson