riscvarchive/riscv-binutils-gdb #264: Is there a way to let gdbserver instruct gdb client to use compressed breakpoints?

lbmeng, 2021-08-31:
I was wondering, is there a way for gdbserver to tell gdb client to use compressed breakpoints by default, ie:

`>>> target remote :1234`

At this point, gdb server tells gdb clint to "set riscv use-compressed-breakpoints on" automatically.

Is this possible? Or enhancement to gdb?

jim-wilson, 2021-08-31:
There aren't any gdb experts here.  Try an FSF gdb mailing list.

However, I would guess that you can expand the XML that the target sends to the client.  Currently it is a list of registers and sizes.  Maybe we could add one more feature for compressed breakpoints, and if the target returns XML for the compressed breakpoint feature, then we can turn it on or off as per the target support.  Targets would have to be extended to include the new XML feature.  I only know of 3 targets: gdbserver, qemu, and openocd.  We can make the XML feature optional in which case if it is missing you get default gdb behavior which is "set riscv use-compressed-breakpoints auto".

This new XML feature could perhaps be more general, and could be an architecture string for instance, where we can then check for 'c' in the architecture string.  Or could just be an indication if we have compressed support, though I don't think we have any other use for compressed in gdb at the moment besides compressed breakpoints.

lbmeng, 2021-09-03:
Thanks Jim. I guess we will wait for someone to implement the gdb changes first.

closed 2021-09-03 by lbmeng