riscvarchive/riscv-binutils-gdb #268: Objdump and auipc: the immediate value should be displayed shifted

ronan-lashermes, 2022-02-02:
# Bug description
Here are some extracts from objdump desassembling (objdump -D). (Using up-to-date riscv-gnu-toolchain that points to riscv-binutils-gdb commit 116a737f438d03a1bd6aa706b6ea0b4022f3b7e2)
```
  48:  00003117          	auipc	sp,0x3
 1dc:  00001197          	auipc	gp,0x1
 1f8:  00001517          	auipc	a0,0x1
```
Since, in auipc, the immediate encodes for bits 31 down to 12. The displayed immediate value would be easier to interpret if left shifted by 12:
```
  48:  00003117          	auipc	sp,0x3000
 1dc:  00001197          	auipc	gp,0x1000
 1f8:  00001517          	auipc	a0,0x1000
```

aswaterman, 2022-02-03:
Not a bug so much as a quirk of the RISC-V assembly language.  Anyway, this change can't be made for practical reasons, because it's backwards incompatible: assembling `auipc sp, 0x3000` would no longer mean the same thing as it used to.

Note that the `auipc` assembly syntax matches `lui`, which in turn matches MIPS's `lui`, which also keeps the argument right-justified.  So, quirky as this is, it's not unique to RISC-V.

ronan-lashermes, 2022-02-03:
Fair enough. Never worked with MIPS before, so that was a new "subtlety" for me with RISC-V.

closed 2022-02-03 by ronan-lashermes