riscvarchive/riscv-binutils-gdb #269: Does anyone know the status about zfh support of binutils?

zhongjuzhe, 2022-03-21:
I have seem this someone commit this patch:https://sourceware.org/pipermail/binutils/2022-January/119276.html. But it seems there is no response. For RVV support in GCC, the zfh is also important. Without ZFH support, The FP16 in vector can not be enabled.

kito-cheng, 2022-03-21:
Actually we need zvfh for FP16 vector operations: https://github.com/riscv/riscv-v-spec/pull/780 , but zfh and zfhmin still pre-requirement for that I think, I could ping Nelson to review that to make some progress :P

zhongjuzhe, 2022-03-21:
> Actually we need zvfh for FP16 vector operations: [riscv/riscv-v-spec#780](https://github.com/riscv/riscv-v-spec/pull/780) , but zfh and zfhmin still pre-requirement for that I think, I could ping Nelson to review that to make some progress :P

OK. I am planning to split a clean version of GCC to upstream which only has RVV intrinsic support When our company FSF has been approved. Do you think I need to remove fp16 vector as well?

kito-cheng, 2022-03-21:
> OK. I am planning to split a clean version of GCC to upstream which only has RVV intrinsic support When our company FSF has been approved. Do you think I need to remove fp16 vector as well?

I would suggest that could a separated patch and come after zfh support upstream.

aswaterman, 2022-03-21:
@kito-cheng FYI, current thinking is that Zvfh will require Zfhmin, and Zvfhmin will require nothing. But this is just a proposal today; it has not been formally decided. https://github.com/riscv/riscv-v-spec/pull/780/files

kito-cheng, 2022-03-21:
Oh, thanks for remind me that, I forgot that since LLVM only implement zvfh for now https://reviews.llvm.org/D121345

closed 2022-05-03 by zhongjuzhe